rust-embedded-community/tm4c-hal #9: Move out the bit-banding code

chrysn, 2018-06-21:
The bb module in this HAL is well usable for other chips, I'd like to suggest splitting it out for easier re-use. (Right now I've copy-pasted it to the [efm32gg-hal](https://crates.io/crates/efm32gg-hal) crate).

Not only is the code generic over various chips, but also for memory regions. The comments to `ref_to_bitband` say it can be used only things slightly above 0x2000_000, but actually works for the peripheral registers at 0x4000_0000 just as well.

I expect the challenge to be the characterization of where the resulting bit-banding crate could be applicable. The most generic document I found that describes this very mechanism is 2.2. Memory Model of the [Cortex-M4 Generic User Guide](http://infocenter.arm.com/help/topic/com.arm.doc.dui0553a/DUI0553A_cortex_m4_dgug.pdf), but it appears to be applicable to M3 (at least EFM32) as well.

thejpster, 2018-06-22:
Humm. So I think how I'd propose to make this generic, is to create a Bitband object which represents each region:
```
pub struct Bitband {
    start_addr: usize,
    bitband_addr: usize,
    len: usize
}

impl Bitband {
    pub unsafe fn new(start: usize, bitband: usize, len: usize) -> Bitband { .. }
    pub fn read_bit<T>(&self, address: *const T, bit: u8) -> bool  { ... }
    ...
}
```

The various XXX-hal implementations (or even cortex-m) can then do the unsafe bit and make you an object that's safe to use. I agree the Cortex-M4 spec gives the two regions, but the size of the first region will depend on the amount of SRAM you have on-chip.

chrysn, 2018-06-23:
With this Bitband struct, constant propagation and LTO would ensure that bitband access is the single-static-address-no-runtime-checks operation I'd want for most cases. Do we want/need to consider use cases with less optimization (eg. by encoding addresses and sizes in the type rather than in struct members)? ("No" is a perfectly good answer just as well, just want to make sure we considered it.)

thejpster, 2018-08-12:
Sorry, this one dropped off my radar. In response to your last comment, that sounds interesting, but I'd add that as an additional variant later and working out how that would work shouldn't hold up the basic case outlined at the top.