rust-embedded-community/tm4c-hal #10: Add PWM Support

liamdiprose, 2018-08-01:
Opening this issue to discuss the addition of PWM support.

I'm a computer engineering student that has been allowed to use Rust for a project in one of my embedded software courses. Being an audio-processing project, I will probably need PWM support.

I am happy to add support for PWM. I have managed to get a very basic demonstration working with the tm4c123x library. My next step is to implement the `PwmPin` trait for every supported pin.

Are there any other points we should take into consideration for PWM pins? 
Keen to hear your input.

thejpster, 2018-08-12:
So I use the pins in PWM mode in github.com/thejpster/monotron. I've been meaning to put together a simple API but for now I do it through raw svd2rust calls.

Whatever we do should implement the embedded-hal traits. What state are they in for PWM? I haven't checked lately.

thejpster, 2018-08-12:
As an aside, Monotron is scheduled to receive a dual-6-bit PWM audio output at some stage, so any help getting that running would be excellent. It has to be six bit because I think the PWM cycles need to divide down nicely into the video line timing, where on Monotron a video line is precisely 2112 clock cycles long (and 2112 = 33 x 2^6). I think that gives 10 effective bits in count-up/down mode.

thejpster, 2018-10-03:
Just to follow up on the above - two points of clarification:

* The TM4C123 has two peripherals which can do PWM - the Timers and the PWM peripheral.
** The Timers only do count-up (which is fine for Monotron's horizontal sync signal).
** The PWM module does count-up-down as well (which is ideal for Monotron's audio output), but doesn't exist on the older LM4F120, only the TM4C12x - limiting backwards compatibility.
* The Monotron audio doesn't need to be dual-6-bit, 8-bit audio works just fine as long as you accept that you can't go full rail (i.e. maximum 256 out of 264) and set your DC bias appropriately.

mokomull, 2020-05-27:
I took a stab in #28, but I can only test on an LM4F120 (= TM4C1233, it seems).  As far as I can tell, none of the tm4c123x have the dedicated PWM peripheral; that appears to be specific to the 129x series.

thejpster, 2020-05-30:
Nice, thanks! Just to note that the TM4C123GH6PM definitely has PWM: https://docs.rs/tm4c123x/0.9.0/tm4c123x/struct.PWM0.html.

The LM4F120 series was canned due to an alarming number of errata. I think the TM4C123G is the closest to the LM4F120, but they added a quadrature decoder and the aforementioned PWM peripheral.

closed 2020-07-19 by thejpster