rust-embedded-community/tm4c-hal #11: Add ADC support

liamdiprose, 2018-08-01:
Opening this issue to discuss the addition of ADC support.

Just as stated in my other issue #10 , I have an upcoming project that aims to process audio using Rust. For this project, I will need ADC support for audio input.

I am happy to add support for the ADC's, but unlike the PWM support, I do not know how much work is required. 

Are there any points we need to discuss concerning the addition of ADC support?

thejpster, 2018-08-12:
As with PWM, if there's a HAL, we should use it. If there isn't a HAL, we should propose one (keeping it in this tree for now, until it is accepted/merged up-stream).

I haven't had need for an ADC so I'm open to suggestions on the API.

thejpster, 2019-11-10:
So, nrf52-hal has an API for their `saadc` peripheral. We could crib that?

liamdiprose, 2019-11-10:
Hey, thanks for pinging me on this. I've just started getting back into Rust / Embedded.

Just so we're clear, by API you're talking about the `Saadc::new` function and `SaadcExt` trait?

vipierozan99, 2021-01-05:
Hey, I`ve done some tinkering with the adc module, it´s very early tho. Suggestions and directions are welcome, there is a pull request #46