rust-embedded-community/tm4c-hal #34: I2C not working anymore

dkm, 2020-07-20:
After last changes in I2C, it seems the logic of the busy_wait macro is not correct.
I've been able to patch it locally and will push a clean proposal tomorrow (@thejpster asked me to file an issue).

Basically, checking for BusBusy everywhere is not correct as it will always be set between START and STOP, so during a transaction, we should not return an error (that's what I'm currently seeing).

Previous code was basically adapted from stm32f30x-hal and I see that the busy_wait macro has been changed (no more flag as parameter) and is not really busy_waiting on anything but the BUSY bit. The macro is now wrapped in `loop{}`. Was is not better to have the busy wait inside the macro ? Maybe there's something I'm not seeing here.

The original code was doing things like:
- wait for `busbsy` to be clear
- do something
- wait for `busy` to be clear
- do something
....

I see that previous code was wrongly looking at all `mcs` bits without waiting for `busy` to be clear, but maybe the cleaning went too far ? At least with my patch that reverts back the logic above, my application is working correctly again.

thejpster, 2020-07-23:
@dtwood any thoughts on this one?

thejpster, 2020-07-30:
@dtwood?

PR #35 (2020-08-10): Fix i2c busy / bus busy handling
Restore previous logic where a transaction starts by waiting for the controler
to be ready and the bus available and then only wait for the controller to be ready
at intermediate step (not the bus to be available as it will be busy).
Modify the i2c_busy_wait to implement this behavior.

fix #34
commit a6555649 (2020-08-10): Fix i2c busy / bus busy handling

Restore previous logic where a transaction starts by waiting for the controler
to be ready and the bus available and then only wait for the controller to be ready
at intermediate step (not the bus to be available as it will be busy).
Modify the i2c_busy_wait to implement this behavior.

fix #34
closed 2020-08-10 by thejpster