rust-embedded-community/tm4c-hal #44: Updates for dslite2svd

thejpster, 2020-12-04:
Hi. Sorry for opening this here, but....

1. Can we turn on the issue tracker on the dslite2svd fork.
2. Can we set up a GitHub Action to build the SVD files
3. Can we delete the SVD files from that repo (I never like auto-generated code in my repos)
4. Can we create a tm4c123-pac repo (and a 129) instead of it being inside the dslite2svd repo
5. Can we do a few rounds of pushing to crates.io and updating Cargo.toml.

I accept the above are strong opinions, but the are only weakly held so feel free to rebut!

eldruin, 2020-12-04:
Oops, sorry about that. You should have had permissions to turn on the issue tracker there, though. Let me know if that is not the case.
I moved this issue to the dslite2svd repository now and turned it into a TODO list. We can follow up the discussion there: https://github.com/rust-embedded-community/dslite2svd/issues/1

thejpster, 2020-12-04:
Ah, maybe. I'm on the Android App and it doesn't have all the features.

eldruin, 2020-12-04:
No worries. I'll close this now, then.

closed 2020-12-04 by eldruin